schmttc/EasyThreeD-K7-STM32 #3: K7 with ATmega1284

tux-friend, 2022-10-24:
Hello
So I have a K7 with ET-4000 V2 board with ATmega 1284. Would this marlin version also work for the atmega?
I found out that the Board version would be Melzi for the ATmega, but do I need to change anything else?

schmttc, 2022-10-24:
With a different board, you may also need an updated src\pins file.
You can email info@easythreed.com and ask them for a copy of the source code for your board. You can then see which board profile they use to compile, and if they have made any changes to the relevant pins file.

tux-friend, 2022-10-28:
Got your Mod V1 ported to K7. Great and really nice work! Thanks!

closed 2022-10-28 by tux-friend

schmttc, 2022-11-01:
Glad it's all working for you! I've linked to your firmware in the README of this project in case anyone is looking for it.