schmttc/EasyThreeD-K7-STM32 #4: K7 with CF-M12 board

andypiper, 2023-01-13:
Any idea whether this would work with a K7 that has a board marked "CF-M12" - the chip is a Nation N326455 Rel 7. Support e-mail has sent me a default firmware binary, but not the source...

schmttc, 2023-01-14:
Hi! Based on [this article](https://www.programmersought.com/article/32348457342/) , the N32 is a replacement for STM32, but is  not 100% software compatible. So maybe the firmware could work. 
What files are on your SD card?  Is there an mkslite.bin, or other .bin file?

Based on some googling, the Voxelab Aquila has been release with the N32 and some other variants. You might be able to use the HAL config from this repo 
 https://github.com/Voxelab-64/N32

Can you send a photo of your mainboard? If it's otherwise the same layout as another board then based on [this](https://www.programmersought.com/article/85117803319/)  I'd take a guess that the pins are the same as the STM32 equivalent board.

You can also keep trying to sweet talk the vendor into sending the source :)

Are there any specific features you want from a new firmware?

andypiper, 2023-01-14:
I’ve had them send me binary of the firmware but they don’t seem to want (?) to share the source - I have a heated bed arriving soon, and also OctoPrint is reporting that host actions are disabled so it would be nice to enable those. It has 2.0.9.1 installed by default.

I will check out the Voxelab you mention, good tip!

[image]
[image]
[image]

[image]

andypiper, 2023-01-14:
They sent me `new_firmware.bin` as the default firmware, with a `CFCONFIG.txt` that doesn’t make complete sense to me. 

```text
#This a test file

>DEFAULT_X_STEPS_PER_UNIT            606
>DEFAULT_Y_STEPS_PER_UNIT            606
>DEFAULT_Z_STEPS_PER_UNIT            1167
>DEFAULT_E_STEPS_PER_UNIT            1040
>DEFAULT_X_MAX_ACCELERATION            300
>DEFAULT_Y_MAX_ACCELERATION            300
>DEFAULT_Z_MAX_ACCELERATION            200
>DEFAULT_E_MAX_ACCELERATION            300
>INVERT_X_DIR            1
>INVERT_Y_DIR            0
>INVERT_Z_DIR            1
>INVERT_E0_DIR            1
>X_HOME_DIR            -1
>Y_HOME_DIR            -1
>Z_HOME_DIR            -1

>X_MAX_FEEDRATE 20
>Y_MAX_FEEDRATE 20
>Z_MAX_FEEDRATE 10
>E_MAX_FEEDRATE 20

>X_MAX_FEEDRATE	20			#unit mm/s
>Y_MAX_FEEDRATE	20			#unit mm/s
>Z_MAX_FEEDRATE	5			#unit mm/s
>E_MAX_FEEDRATE	20			#unit mm/s

>DEFAULT_XJERK  2.0
>DEFAULT_YJERK  2.0
>DEFAULT_ZJERK  0.4
>DEFAULT_EJERK  0.5
```

andypiper, 2023-01-20:
I've now installed an integrated heated bed with an alternative firmware that they supplied (binary, no source)

Output from Octoprint / Marlin connection:

```
Changing monitoring state from "Connecting" to "Operational"
Recv: ok
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin 2.0.9.1 (May 16 2022 18:25:11) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:CF-M12 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_POS:0
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:0
Recv: Cap:RUNOUT:0
Recv: Cap:Z_PROBE:0
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:HOST_ACTION_COMMANDS:0
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:SDCARD:1
Recv: Cap:REPEAT:0
Recv: Cap:SD_WRITE:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:1
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: Cap:COOLER_TEMPERATURE:0
Recv: Cap:MEATPACK:0
Recv: ok
Send: M21
Recv: echo:SD card ok
Recv: ok
Send: M155 S2
Recv: ok
Send: M20
Recv: Begin file list
Recv: ROCKET~1.GCO 1343370
Recv: End file list
Recv: ok
Recv:  T:20.37 /0.00 B:19.26 /0.00 @:0 B@:0
Recv:  T:20.33 /0.00 B:19.26 /0.00 @:0 B@:0
Recv:  T:20.21 /0.00 B:19.34 /0.00 @:0 B@:0
Recv:  T:20.18 /0.00 B:19.30 /0.00 @:0 B@:0
Recv:  T:20.62 /0.00 B:19.12 /0.00 @:0 B@:0
Recv:  T:20.23 /0.00 B:19.26 /0.00 @:0 B@:0
Recv:  T:20.55 /0.00 B:19.28 /0.00 @:0 B@:0
```

Again this one doesn't have host commands enabled. This was installed by putting the binary and the CFCONFIG.txt file on the SD card and powering on, waiting for ~1 min, and it seemed to update.

The CFCONFIG is different to the default:

```text

# Travel limits after homing (units are in mm)
>X_MIN_POS 0
>Y_MIN_POS 0
>Z_MIN_POS 0
>X_MAX_POS 100
>Y_MAX_POS 100
>Z_MAX_POS 100

#============================== Stepper Motor Settings ============================

#Invert the stepper direction.
>INVERT_X_DIR 	                1	
>INVERT_Y_DIR 	                0
>INVERT_Z_DIR                	1
>INVERT_E0_DIR           	1

#Movement setting
>DEFAULT_X_STEPS_PER_UNIT	606	#Default Axis-X Steps Per Unit (steps/mm)	
>DEFAULT_Y_STEPS_PER_UNIT	606	#Default Axis-Y Steps Per Unit (steps/mm)	
>DEFAULT_Z_STEPS_PER_UNIT	1167	#Default Axis-Z Steps Per Unit (steps/mm)	
>DEFAULT_E_STEPS_PER_UNIT	1040	#Default Axis-E Steps Per Unit (steps/mm)	

>X_MAX_FEEDRATE		20	#Default Axis-X Max Feed Rate (mm/s)		
>Y_MAX_FEEDRATE		20	#Default Axis-Y Max Feed Rate (mm/s)		
>Z_MAX_FEEDRATE		10	#Default Axis-Z Max Feed Rate (mm/s)		
>E_MAX_FEEDRATE		70	#Default Axis-E Max Feed Rate (mm/s)		

>DEFAULT_ACCELERATION		100	#X, Y, Z and E acceleration for printing moves		
>DEFAULT_RETRACT_ACCELERATION	100	#X, Y, Z and E acceleration for retracts		
>DEFAULT_TRAVEL_ACCELERATION	100	#X, Y, Z acceleration for travel (non printing) moves	
>DEFAULT_MINIMUMFEEDRATE	0.0	#minimum feedrate						
>DEFAULT_MINSEGMENTTIME		20000	#minimum time in microseconds that a movement needs to take if the buffer is emptied. 
>DEFAULT_MINTRAVELFEEDRATE	0.0	#		

>DEFAULT_XJERK 			2.0	#Default Axis-X Jerk (mm/s)	
>DEFAULT_YJERK 			2.0	#Default Axis-Y Jerk (mm/s)	
>DEFAULT_ZJERK 			0.4	#Default Axis-Z Jerk (mm/s)	
>DEFAULT_EJERK 			0.5	#Default Axis-E Jerk (mm/s)	

#============================== Endstop  Settings ===========================

>MIN_SOFTWARE_ENDSTOPS	 1              # 0:axes can move below MIN_POS;  1:axes won't move below MIN_POS.
>MAX_SOFTWARE_ENDSTOPS 	 1              # 0:axes can move below  MAX_POS;  1:axes won't move below MIN_POS.

# Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
>X_MIN_ENDSTOP_INVERTING 1		# set to true to invert the logic of the endstop.
>Y_MIN_ENDSTOP_INVERTING 1		# set to true to invert the logic of the endstop.
>Z_MIN_ENDSTOP_INVERTING 1		# set to true to invert the logic of the endstop.
>X_MAX_ENDSTOP_INVERTING 0		# set to true to invert the logic of the endstop.
>Y_MAX_ENDSTOP_INVERTING 0		# set to true to invert the logic of the endstop.
>Z_MAX_ENDSTOP_INVERTING 0		# set to true to invert the logic of the endstop.

#==============================  Home Settings ==============================

>X_HOME_ENABLE 1
>Y_HOME_ENABLE 1
>Z_HOME_ENABLE 1

>X_HOME_DIR  -1 				# Direction of endstops when homing; 1=MAX, -1=MIN :[-1,1]
>Y_HOME_DIR  -1				# Direction of endstops when homing; 1=MAX, -1=MIN :[-1,1]
>Z_HOME_DIR  -1				# Direction of endstops when homing; 1=MAX, -1=MIN :[-1,1]
>HOMING_FEEDRATE_XY 2400                # Homing X Y speeds (mm/m)
>HOMING_FEEDRATE_Z  300                 # Homing Z speeds (mm/m)
>HOME_Y_BEFORE_X	0       	# When G28 is called,0: X home before Y; 1: Y home before X

```

I don't think this is actually a compile-time set of configuration options, though? otherwise why supply it as a text file?

In terms of the board, I took a bit more of a look when I opened the printer to plug in the heated bed. Based on layout and photos it looks closer to the ET4000 than to the ET4000+ (positioning of ports etc). Difficult to take a clear image as the cables are hot glued to the board in places!

[image]

[image]

I've yet to try a Marlin build of my own.

schmttc, 2023-01-20:
A couple more background items
- https://github.com/alexqzd/Marlin/issues/11
- https://github.com/MarlinFirmware/Marlin/pull/22286
- https://www.reddit.com/r/VoxelabAquila/comments/pn8s6s/help_modifying_alexs_marlin_fw_for_e3d_v6/
- Different board version images, it looks like the CF-M12 is different https://m.aliexpress.com/i/4001148621529.html

I'd suggest  issuing the M43 command to marlin to get the pinout list, which will hopefully match an existing STM32 board, and use the platform.io config from Alex's firmware to roll your own. 

I'm afraid I can't be of much more help.

andypiper, 2023-01-20:
that's cool, thank you so much!

andypiper, 2023-01-23:
M43 unfortunately only works if the board has debugging enabled and is disabled in the production builds, so that is not working for me, but you've been hugely helpful here in educating me and providing pointers. Thank you.

closed 2023-02-05 by schmttc

andypiper, 2023-04-27:
Hello again! Just wanted to let you know that I have ended up acquiring an ET-4000+ board for the printer, and that I'm now rocking your firmware on it, which is far more satisfying than the previous situation (I can now enable the options I actually want).

Annoyed that EasyThreed didn't share the source for their modifications with the CF-M12 board and chipset, but thank you for your patience with my previous questions, and I appreciate the work you've done to make recent Marlin work here.

Out of curiousity, have you tried the input shaping options at all? I may have a look at those now I'm more free to use the printer as I want!

andypiper, 2023-04-27:
... also I'm curious about the BL Touch mod you posted on Thingiverse - specifically the board wiring part. Should that be obvious in terms of where it plugs in?

schmttc, 2023-04-28:
- I haven't tried input shaping yet, but I'm interested to see how it goes!
- It's on my backlog to document the BLTouch configuration, but in the image:
  - Endstop and Ground pins go to Z+ (black+white)
  - Servo GND and 5V to MT_DET (brown+red)
  - Servo Signal to the far right pin on the green header row (orange)
  - And the pin config should be in the '3dtouch' branch