schmttc/EasyThreeD-K7-STM32 #7: How to use a RepRapDiscount Full Graphic Smart Controller?

btlucas, 2023-02-24:
So, I'm trying to figure out how to adapt the firmware to use a [RepRapDiscount Full Graphic Smart Controller](https://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller), going through what I could find I tried modifying the pins_MKS_ROBIN_LITE.h to use the LCD panel and disabled the EASYTHREED_UI on the configuration.h file. The build run was successful and the flashing seemed to be successful too because the mksLite.bin was correctly changed to .cur but the LCD is not powering up.
I used the datasheet you provided at https://github.com/schmttc/EasyThreeD-K7-STM32/blob/Additional-Files/ET4000%2B.pdf
Does anyone know how to do this configuration or have an example of adding a different LCD screen?
I've seen videos of the EeasyThreed X4 that has an LCD screen and uses the exact same board (ET-4000+) so I know that is possible.

btlucas, 2023-02-24:
The ET-4000 v2 uses this layout to connect to the RepRap display 

[image]

I want to do the same but with the ET-4000+ instead, do you @schmttc or anyone else here knows the pin layout for that?

schmttc, 2023-03-08:
You're on the right track, you have to map the schematic diagram you noted and make sure the wiring aligns to the LCD board. I don't have an LCD so can't assist further.
Here's another repo for the V2 board and a different LCD https://github.com/tux-friend/Easythreed-K7-ET-4000-V2

Marking this issue as closed, as it is not specifically relevant to the ET-4000+ board.

closed 2023-03-08 by schmttc