schmttc/EasyThreeD-K7-STM32 #8: Increase Bed Size on K7

Skilledcamman, 2023-04-22:
I was thinking of buying this 3d printer from Aliexpress ([TISHRIC k7](https://www.aliexpress.com/item/1005004426470257.html?spm=a2g0o.cart.0.0.8efa38davELaKq&mp=1) 
and I wanted to increase the bed size do you think I can do that if yes then how

schmttc, 2023-04-22:
1. Theoretically you can increase the bed size, but I strongly advise against it, as the latest models of K7 are coming with a different CPU that is not compatible with any custom firmware yet. This also means you cannot remove the stepper motor backlash via firmware.
2. An entry level full sized 3D printer like Ender 3 doesn't cost much more money and is in all aspects superior - has a larger print bed, no backlash, is faster, more reliable, can replace all the parts, has a heatedbed, etc.

Jerryjzr, 2023-05-10:
> 1. 从理论上讲，您可以增加床的尺寸，但我强烈建议不要这样做，因为最新型号的 K7 配备了不同的 CPU，该 CPU 尚不兼容任何自定义固件。这也意味着您无法通过固件去除步进电机齿隙。
> 2. 像Ender 3这样的入门级全尺寸3D打印机不会花费更多的钱，并且在各个方面都更胜一筹 - 具有更大的打印床，没有反弹，更快，更可靠，可以更换所有部件，具有加热床等。
I bought a k7 3 days ago, his processor is shown in the picture
[image]
The model is AT32F403ARCT7
According to the official website of this chip, this chip has high compatibility with the stm32f103rct6 originally used by k7. According to the documentation provided by the chip supplier, the program of stmf103 can be directly run on AT32F403ARCT7

schmttc, 2023-05-10:
Hi @Jerryjzr , that is interesting. 
Another customer got a different board type CF-M12 with a N32 processor https://github.com/schmttc/EasyThreeD-K7-STM32/issues/4

If the board i

Jerryjzr, 2023-05-10:
> 嗨，这很有趣。
> 
> 另一位客户获得了带有 N12 处理器 #32 的不同主板类型 CF-M4
> 
> 如果您的主板仍然是 ET4000+ 并且具有兼容的处理器，则此固件应该仍然可以工作。如果您尝试，请告诉我们结果！
I flashed this program on my printer, it can run but it will crash, it may be the reason for the heating table, because my machine is not equipped with a heating table, and then I replaced it with you "Branch: Original Firmware Marlin 1.1.1", and it worked perfectly.

Jerryjzr, 2023-05-10:
[image]
This is the chip specification comparison table given by the chip supplier

Jerryjzr, 2023-05-10:
> 嗨，这很有趣。
> 
> 另一位客户获得了带有 N12 处理器 #32 的不同主板类型 CF-M4
> 
> 如果您的主板仍然是 ET4000+ 并且具有兼容的处理器，则此固件应该仍然可以工作。如果您尝试，请告诉我们结果！
Thank you for the program it helped me a lot
Now I turn off the "heated bed" option in the slicing software and print the job, and it works perfectly. No more freezes, I'm now trying to get this k7 to run "klipper"

schmttc, 2023-05-10:
@Jerryjzr I was interested in trying Kipper, but have not had the time to try.
I believe this firmware should work for the ET4000+ board https://github.com/makerbase-mks/Klipper-for-MKS-Boards/tree/main/MKS%20Robin%20Lite%20V1.1
Please let me know how it goes!

Jerryjzr, 2023-05-11:
> @Jerryjzr I was interested in trying Kipper, but have not had the time to try. I believe this firmware should work for the ET4000+ board https://github.com/makerbase-mks/Klipper-for-MKS-Boards/tree/main/MKS%20Robin%20Lite%20V1.1 Please let me know how it goes!

I ran the "bin" provided by mks on my printer, but it doesn't seem to work properly, "Shutdown due to webhooks request" appears in the web interface
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
Config, and restart the host software.
Printer is shutdown "Maybe the versions don't match, I'm trying more

schmttc, 2023-08-03:
Closing ticket as original issue is addressed

closed 2023-08-03 by schmttc