schmttc/EasyThreeD-K7-STM32 #10: EasyThreed K9 version request

JohnTheCoolingFan, 2023-09-18:
I have an EasyThreed K9 with ET4000+ board, but it has a few differences. Some perfipherals are missing and wifi board connector is used by the front panel buttons. Would you be interested in making another repo for K9 if I provide photos and pin/button mappings?

schmttc, 2023-09-20:
Hi, I have been thinking that it would make sense to have a central repo to manage the different device firmwares. 
I need to think about the best way to organise this to keep maintenace low, but happy to provide a branch you can PR to in the mean time? I don't have time at the moment to do any development work.

JohnTheCoolingFan, 2023-09-20:
Yeah, both of thsoe are great ideas. I'll do what I can in the meantime. I was planning to mod my printer anyways. Thanks for the response!

schmttc, 2023-09-20:
https://github.com/schmttc/EasyThreeD-K7-STM32/tree/ET4000PLUS-K9

I've put a few pointers in the README to get started, check the master branch READMA for additional details, and the Additional-Files branch for board schematics

JohnTheCoolingFan, 2023-09-29:
Finally got to have a look. From what I can see, controls are mostly the same but I think the 4 front leveling buttons need to be added. I'll check the pins later. Also will take a better look at what STM32 model is installed in my printer.

schmttc, 2023-10-23:
@JohnTheCoolingFan have you/can you email Easythreed support and ask for a copy of the source code for your printer? They were a bit hesitant for me but it would make the development work a lot easier.

JohnTheCoolingFan, 2023-10-23:
That's an interesting suggestion. Probably worth a try.
I'm a bit concerned whethr this printer model is even real as it's not displayed at easythreed website.

schmttc, 2023-11-19:
I was just thinking about this and remembered that all of the EasyThreed printers are all the same hardware in a different form factor; so you could try and  flash my K7 firmware and it might all just work, except for the homing buttons.
If it doesn't work, you can just flash back,I've flashed plenty of bad firmwares and always been able to roll back.
Try this one https://github.com/schmttc/EasyThreeD-K7-STM32/blob/backlashfix/mksLite.bin

schmttc, 2023-12-01:
I've started creating the central repo to firmware for multiple printers, if you have any feedback please let me know https://github.com/EasyThreed-Community-Firmware/Marlin

Ajimaru, 2024-03-05:
Which Firmware can I use for testing on a K9?

JohnTheCoolingFan, 2024-03-06:
Are you asking for a compiled file @Ajimaru? I can compile it if you want.

JohnTheCoolingFan, 2024-03-07:
Here you go, uploaded it to my web server: https://jtcf.ru/EasyThreed-K9-firmware/mksLite-c4def742.bin

Ajimaru, 2024-03-07:
> Here you go, uploaded it to my web server: https://jtcf.ru/EasyThreed-K9-firmware/mksLite-c4def742.bin

I have installed this firmware via the sd-card.
```
Firmware Name: Marlin 2.1.2.1 (Mar 7 2024 22:21:12)
Extruder Count: 1
Machine Type: 3D Printer
Protocol Version: 1.0
Source Code Url: github.com/MarlinFirmware/Marlin
Uuid: cede2a2f-41a2-4748-9b12-c55c62f367ff 
``` 

Good news, headbet is working.
Bad news, the four bed leveling buttons and the home button on the printer does not work anymore.
COntroling via USB does not work either.
In addition all theese settings are on zero.
```
[Steps]
[Feedrate]
[Acceleration]
[Offsets]
[PID]
[Material]
[Linear Advance]
[Advanced]
```

JohnTheCoolingFan, 2024-03-07:
That's weird. Can you open the printer and read a model of the board to make sure it is ET4000+?

Photos of the board and the extension board on the place of the wifi module would be nice for finding the culprit too.

Ajimaru, 2024-03-08:
Yes the board is an ET4000+ , here are some pictures.

[image]
[image]
[image]
[image]
[image]

I have set the missing values according to [EasyThreed K9 Marlin factory firmware configuration ](https://gist.github.com/JohnTheCoolingFan/39227f4ddd12a7c614d4e360d0473e6a)

**M503** shows this, after powercylce of the printer, the values are back to zero.

```
echo:; Linear Units:
echo:  G21 ; (mm)
echo:; Temperature Units:
echo:  M149 C ; Units in Celsius
echo:; Filament settings (Disabled):
echo:  M200 S0 D0.00
echo:; Steps per unit:
echo:  M92 X606.00 Y606.00 Z600.00 E1040.00
echo:; Max feedrates (units/s):
echo:  M203 X40.00 Y40.00 Z10.00 E70.00
echo:; Max Acceleration (units/s2):
echo:  M201 X100.00 Y100.00 Z50.00 E100.00
echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
echo:  M204 P100.00 R100.00 T100.00
echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
echo:  M205 B0.00 S0.00 T0.00 J0.00
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Hotend PID:
echo:  M301 P22.20 I1.08 D114.00
echo:; Linear Advance:
echo:  M900 K0.00
echo:; Stepper motor currents:
echo:  M907 X0 Z0 E0
echo:; Backlash compensation:
echo:  M425 F0.00 S0.00 X0.00 Y0.00 Z0.00
``` 

Still only nozzle and bed heating works, all functions related to motors are dead.
You have the stock firmware?

JohnTheCoolingFan, 2024-03-08:
Yes, I have the stock firmware which I got from the manufacturer. I also uploaded it to my webserver: http://jtcf.ru/EasyThreed-K9-firmware/EasyThreed-K9-factory-firmware.bin

JohnTheCoolingFan, 2024-03-08:
Your daughter board for the front levelling buttons looks different. Can you please use a multimeter to map the contacts on the daughter board on the front to the contacts on the board in the wifi module slot?
[image]

Ajimaru, 2024-03-10:
> Yes, I have the stock firmware which I got from the manufacturer. I also uploaded it to my webserver: http://jtcf.ru/EasyThreed-K9-firmware/EasyThreed-K9-factory-firmware.bin

The Printer is working again with the factory firmware!

Ajimaru, 2024-03-10:
> Your daughter board for the front levelling buttons looks different. Can you please use a multimeter to map the contacts on the daughter board on the front to the contacts on the board in the wifi module slot? ![pins_to_map](https://private-user-images.githubusercontent.com/43478602/311383241-d8370af4-aeb5-42bc-817f-eececfeabd65.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAwODAxMjksIm5iZiI6MTcxMDA3OTgyOSwicGF0aCI6Ii80MzQ3ODYwMi8zMTEzODMyNDEtZDgzNzBhZjQtYWViNS00MmJjLTgxN2YtZWVjZWNmZWFiZDY1LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzEwVDE0MTAyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI5YzZhNjU4Zjc4MDRmZDFkYjI1NmE4Y2ZiNzY3ODFjYzczMDI0ZWI1OTAzMjQ3NDQ1ZDY2MWJmNjEzY2UyMzAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Iu3tZoi7uCHY5V_VlV337mvN6OipQq0Xtk7ExCxn8RA)

Here are the mapped contact:

| Pin U3 | Pin U4 |
| ---------- | ---------- |
| 1 | 1 |
| 2 | N.C. |
| 3 | N.C. |
| 4 | 3 |
| 5 | 4 |
| 6 | N. C. |
| 7 | 5 |
| 8 | 6 |
| 9 | 2 |
| 10 | N. C. |
| 11 | N. C. |
| 12 | N. C. |
| 13 | N. C. |
| 14 | N. C. |
| 15 | N. C. |
| 16 | N. C. |

[image]

Ajimaru, 2024-03-11:
> Hmm, the buttons are mapped the same way. I don't know what the cause of the issue is and I'm out of ideas, sorry

I may have to copy the **lite_cfg.txt** also on the SD-Card?

JohnTheCoolingFan, 2024-03-11:
I don't know, my firmware worked without issues. Sorry, but I can't help much without actual hardware that has the issue to experiment with.

schmttc, 2024-03-12:
@Ajimaru I've compiled the code from the K9 branch here and the file size is different to the file John uploaded (134KB vs 118KB)
Can you try this one? (remove the .txt when you put it on the SD card)
[mksLite.bin.txt](https://github.com/schmttc/EasyThreeD-K7-STM32/files/14570021/mksLite.bin.txt)

lite_cfg.txt doesn't do anything in this version of the firmware.

Ajimaru, 2024-03-16:
> @Ajimaru I've compiled the code from the K9 branch here and the file size is different to the file John uploaded (134KB vs 118KB) Can you try this one? (remove the .txt when you put it on the SD [...]

@schmttc with your compiled bin file the leveling buttons on the K9 are working.
To make some things clear, I use Octoprint to control the printer.
With the factory firmware Octoprint can control all functions of the K9.
But with your compiled Marlin Firmware only heating the Nozzle and the Bed is working.
Moving the X, Y & Z axis does not work.

I would love to the the K9 fully functional with Marlin to make use of the heat bed.

schmttc, 2024-03-19:
> @schmttc with your compiled bin file the leveling buttons on the K9 are working. To make some things clear, I use Octoprint to control the printer. With the factory firmware Octoprint can control [...]

If the levelling buttons are working, then the motion steppers are configured OK. I wonder if there is an issue between Octoprint and some setting in the firmware.

@Ajimaru 
1. "Moving the X, Y & Z axis does not work." is this when manually controlling motion with the Octoprint GUI?
2. Does motion work when running a print job from gcode?

Ajimaru, 2024-03-19:
> If the levelling buttons are working, then the motion steppers are configured OK. I wonder if there is an issue between Octoprint and some setting in the firmware.
>
> @Ajimaru
>
>     1. "Moving the X, Y [...]

@schmttc I have found the rootcause, after update the firmware a file named **EEPROM.DAT** was created on the SD Card.
To use the Marlin firmware an SD Card with this file on it need to inserted.

schmttc, 2024-03-19:
> @schmttc I have found the rootcause, after update the firmware a file named **EEPROM.DAT** was created on the SD Card. To use the Marlin firmware an SD Card with this file on it need to inserted.

Thanks for your persistence. I have an outstanding action to implement EEPROM correctly #5

Ajimaru, 2024-03-19:
Well I should have a closer look to the closed issues next time.
Besides the EEPROM everything is working as expected, thanks for the work.

novadaemon1, 2024-04-18:
> @Ajimaru I have enabled the onboard EEPROM and uploaded the new binary to the K9 branch [9338b48](https://github.com/schmttc/EasyThreeD-K7-STM32/commit/9338b4863de90db6eae2fcedd507d36d76772c6b), if you try it please let me know the results

Just wanted to let you know that the firmware works with the buttons just fine, but it has some quirks. For example, Z axis homing takes way too long and grinds the motor significantly more than stock firmware. Once I figure out all the issues I will let you know.

Ajimaru, 2024-04-19:
> @Ajimaru I have enabled the onboard EEPROM and uploaded the new binary to the K9 branch [9338b48](https://github.com/schmttc/EasyThreeD-K7-STM32/commit/9338b4863de90db6eae2fcedd507d36d76772c6b), if you try it please let me know the results

I can confirm that this firmware is working now with out the EEPROM.DAT file spresent on the SD-Card.
I also noticed the grinding noise when the Z-Motor has reached the physical endstop.
This behaviour is not present in the stock firmware.

As a request can you please add [Host Action Commands](https://marlinfw.org/docs/configuration/configuration.html#host-action-commands) ?

JohnTheCoolingFan, 2024-05-05:
> I also noticed the grinding noise when the Z-Motor has reached the physical endstop.

I have noticed something similar, which is that it twitches when reaches the physical end stop, trying to step further and I think skipping steps/sequences of steps because of that. The original firmware might have had some mechanism to detect when a motor is not moving the axis any further, or they just drive it differently so that it reaches the end smoothly.

Speaking of motors: I have noticed that my motors are heating up a lot even at idle! I didn't test with the stock firmware, though. The screw that holds the cam in place is very hot, as is the bracket that holds down the motor. Is it a common thing in 3d printing? Should I be worried about damage to motors or parts of the printer?

JohnTheCoolingFan, 2024-05-05:
Ah, and another note, regarding the main button behavior. Holding it down when printing is in progress doesn't stop the print immediately, but probably stops the queue from filling up, a number of operations sill still execute. The sock firmware stops the print immediately.

Also, after the print is finished, the button is still flashing like it's in printing mode, but it had already finished the routine (homed x and y axes). Holding down the button would stop the print and then revert to the default behavior of moving the print head 1cm up when held down for 3 seconds.

schmttc, 2024-05-08:
1. @Ajimaru I'll get host action commands added to the binary soon.

2. Can one of you please upload a copy of the original lite_cfg.txt, I have a theory about the homing and heat issues, maybe the vendor has reduced the current, these steppers are only rated for around 250ma and this firmware provides 1A. I didn't notice an issue on the K7 but the steppers are enclosed.
Less current means less heat (steppers are always 'on' even when not moving); and less torque so the motor will stall more easily as the end of homing and not crunch as the belt slips.

Regarding button behaviour
3. The Marlin devs didn't like the hard stop when I merged into mainline, but happy to change it back because it makes more sense and replicates out of box behaviour.
4. Button still flashing after print sounds like a bug, can fix the state

schmttc, 2024-05-09:
@Ajimaru @JohnTheCoolingFan 
Here's a binary to test with items 1,2 and 3 addressed plus 5.  speed increase on Z homing , can you let me know if it works or what doesn't?
4. needs more time than I have at the moment
 
[mksLite.bin.txt](https://github.com/schmttc/EasyThreeD-K7-STM32/files/15258067/mksLite.bin.txt)

Ajimaru, 2024-05-09:
> 1. @Ajimaru I'll get host action commands added to the binary soon.
>
>     2. Can one of you please upload a copy of the original lite_cfg.txt, I have a theory about the homing and heat issues, maybe [...]

@schmttc here is the factory [lite_cfg.txt](https://github.com/schmttc/EasyThreeD-K7-STM32/files/15258391/lite_cfg.txt)

Ajimaru, 2024-05-09:
> 1. @Ajimaru I'll get host action commands added to the binary soon.
>
>     2. Can one of you please upload a copy of the original lite_cfg.txt, I have a theory about the homing and heat issues, maybe [...]

I can confirm that 1. & 2. are fixed, good job tahnk you!
3. & 4. I can not test because I use Octoprint to control the printer.

schmttc, 2024-05-09:
Thanks for the file and confirming the results @Ajimaru 

The stepper current is 750/800ma in your config file, and I've reduced it to 500 from 1000, which is still double the rated 200ma so it should be fine. But if you get layer shifting problems or it feels like the printer is skipping steps, we can return it to the default setting.

These changes are in the 'k9-fixes' branch, I'll merge it with ET4000PLUS-K9 once the other changes are confirmed.

JohnTheCoolingFan, 2024-05-09:
> @Ajimaru @JohnTheCoolingFan Here's a binary to test with items 1,2 and 3 addressed plus 5. speed increase on Z homing , can you let me know if it works or what doesn't? 4. needs more time than I [...]

Trying out this firmware right now, immediately can say that it is much more quiet, no noise when reaching ends. The motors are not heating up, but I'm wondering how that would affect performance (skipped steps, layer shifts, lesser acceleration, printing speed). I like how fast the printer is with the custom firmware compared to stock. Doing a test print  now, will post an update later

schmttc, 2024-05-09:
> Trying out this firmware right now, immediately can say that it is much more quiet, no noise when reaching ends. The motors are not heating up, but I'm wondering how that would affect performance [...]
[mksLite.bin.txt](https://github.com/schmttc/EasyThreeD-K7-STM32/files/15268414/mksLite.bin.txt)

Here's one with the out of box current settings you can try as well. Let me know if the homing is still smooth and the heat OK, hopefully this is the right balance.

JohnTheCoolingFan, 2024-05-23:
The motors heat is OK, getting a bit warm to the touch and just that. I haven't noticed layer shifts or skipped steps, only some backlash in the axes, which can be seen on a benchy, for example. In the compartment behind the cabin the wall might have a consistent gap between it and the edge of the top surface [image]
Accounting for that would probably require klipper, but  it's not a big issue for my prints.

JohnTheCoolingFan, 2024-05-23:
There  is an issue with the interface though: holding down the button for 3 seconds while printing stops the print, but after that the printer does not react to the action (trying to move the print head up)

schmttc, 2024-05-24:
Thanks @JohnTheCoolingFan 

Regarding backlash, you can set backlash compensation in Marlin with [M425](https://marlinfw.org/docs/gcode/M425.html). The Klipper folk don't do backlash compensation, their stance is that it should be resolved in the hardware, which is not possible with these printers.
I use this tower to tune backlash settings https://www.thingiverse.com/thing:2371069

I'll look into the interface issue over the next week or two as I find time, will compile for my K7 so I can test before sending it over.
If you want to use this firmware until then I suggest a quick power on/off between prints.

JohnTheCoolingFan, 2024-05-26:
@schmttc thanks for the tools for backlash compensationm, but I've re-encountered  https://github.com/MarlinFirmware/Marlin/issues/23741 when  I  added the gcode, looks like you had or still have this problem too.
[image]
[image]

JohnTheCoolingFan, 2024-05-26:
Ah, this is tracked in #2

JohnTheCoolingFan, 2024-05-26:
Update on backlash compensation: I've managed to merge `backlashfix` into `k9-fixes` and compile that, which resolved the issue. I have calibrated my printer and now I use a 0.32mm backlash compensation value for all axes, as they are completely identical. I can probably put that into the firmware, but I don't want to do that right now. I still  have a problem of seemingly squished first few layers.
The circles have become noticeably more round and less square, which is very nice to see. Still, a 3mm hole turned out to be 2.75mm on average.

schmttc, 2024-05-28:
Great! 
Regarding squashed first layers, try adding some start gcode to your slicer that raises the head say 0.2mm (or whatever you determine)
And for slightly small/large holes, try tweaking Horizontal Expansion (at least that's what it's called in Cura)

JohnTheCoolingFan, 2024-05-30:
And regarding squashed first layers, I think I am noticing some **scratching** on my bed surface, even though I leveled it.

JohnTheCoolingFan, 2024-05-30:
And here's the gcode file I've been using: 
[CFFFP_3DBenchy.gcode.txt](https://github.com/schmttc/EasyThreeD-K7-STM32/files/15502920/CFFFP_3DBenchy.gcode.txt)

schmttc, 2024-05-30:
**Layer Shift**
Looking back at #2 , this could be a more complex issue that's not fully resolved.
1. It's hard to tell where the shifting occurs from that photo, can you change the angle to more side on?
2. Originally the layer shift was only occurring sometimes, but was linked to acceleration. 100mm/s/s always caused shift, 20mm/s/s always didn't. But for my smaller tests the current code fixed the issue. What acceleration and speed are you using? Have you overridden the max speeds and acceleration in your firmware? Can you provide a link to the current code you're using?
3. The shift would show up soon for smaller layer - I see you're using 0.1 from your gcode. Can you try 0.2 layers and see if the shift shows up later? 
4. Comparing the code I worked on with Tom and what was merged back into Marlin, he made some changes on his end afterwards https://github.com/MarlinFirmware/Marlin/pull/26392
So it might be worth trying that.

**Squashed Layers**
I think the backlash comp should be pushing the head up not further down; but try adding some start gcode to lift the head by say 0.4mm

JohnTheCoolingFan, 2024-05-31:
I've re-leveled the bed at 60C and now trying to print benchy with 0.2mm layer height. The heat from the hotbed causes the bottom plastic bracket to warp! That may be the cause of squashed layers, I'll have to see.

Regarding layer shifts, here is a better picture: 
[image]
I've already broken the piece up into small pieces for 3d printed waste storage, can't take more pictures.
I can probably try with different speed, but only on 0.2mm layers, because even with the current speeds it takes much more time than cura predicts.

JohnTheCoolingFan, 2024-05-31:
The code I use to compile the firmware is located here: https://github.com/JohnTheCoolingFan/EasyThreeD-K7-STM32/tree/k9-fixes-backlashfix

JohnTheCoolingFan, 2024-05-31:
The 0.2mm layer benchy printed without any issues.

JohnTheCoolingFan, 2024-06-05:
I've printed a few tests  for bed leveling and discovered that yes, my bed is too close to the nozzle. A layer 1 line had a width of 0.82mm and height of 0.16mm. I'll use a Z offset through gcode first, then will think on more permanent ways of avoiding that problem.
Oh, and the 10mm squares in the bed leveling calibration shape turned out as 10.49-10.72 squares.

JohnTheCoolingFan, 2024-07-07:
I've completed leveling for my 3d printer, still needs adjustments from time to time. But now I'm having an issue where the motors stall at some moves, `G0` or `G1`, just a segment. Causing the whole print to shift. I've moved the backlash to firmware and rebased everything to exclude `backlashfix`. Now I'm upping the current from 0.5A to 0.8A in hopes that it will help with motors stalling. I'm planning to do an upgrade from the channel Made With Layers and currently trying to print reverse-engineered modified split shell inserts.

Rixxxi, 2024-07-08:
Hi guys,
I'm tinkering on my son's EasyThreeD X1 with ET-4000 V2 mainboard and 264-motors on X and Y axis (set to 607 and 605 steps/mm). Mid last year I compiles a firmware with backlash compensation, was suffering from layer shifts and gave up on it since then. 

Last week, after reading through [Layer Shifts issue #2](https://github.com/schmttc/EasyThreeD-K7-STM32/issues/2) I compiled [Marlin 2.1.2.4](https://github.com/MarlinFirmware/Marlin/releases/tag/2.1.2.4) with my own created Configuration.h and Configuration_adv.h (attached). To wrap it up: My configs have backlash compensation, linear advance as well as a LCD-display + encoder enabled (#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER -> https://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller ). 
My understandig is that 2.1.2.4 should have a the fix for the Layer Shifts issue #2 included. Is that correct?

I calibrated the backlash using the [backlash tower from Thingiverse](https://www.thingiverse.com/thing:2371069) and this values gave me nice results: #define BACKLASH_DISTANCE_MM { 0.435, 0.63, 0.255 }
I have not enabled BACKLASH_SMOOTHING_MM so far. Is it recommended to do so?

Already last year in my tests more than 30mm/s speed was causing issues and since then I'm running the printer on DEFAULT_MAX_FEEDRATE { 30, 30, 15, 50 }. Is anyone runnig the 264-motors on 40mm/s without issues ?

For acceleration I didn't make too detailed tests on our printer, but since I read somewhere that 300mm/s² should not be exceeded I compiled with #define DEFAULT_MAX_ACCELERATION { 150, 150, 100, 150 } to be on the safe side.
Lately we finally started a [bigger print](https://www.thingiverse.com/thing:5165613) (downscaled to about 48%) for my son with a lot of details in it. And we were suffering layershifts again :-( 
I adjusted the acceleration down to 100mm/s² for X and Y but still got layer shifts. I gave up on it again.
Now I read this:
>     2. Originally the layer shift was only occurring sometimes, but was linked to acceleration. 100mm/s/s always caused shift, 20mm/s/s always didn't. But for my smaller tests the current code fixed the issue. 
Can you explain a little more about what you found out on the acceleration? Which acceleration settings work, which don't? Is it simple as "lower acceleration is always more safe then higher" or are there particular acceleration settings that will/won't work?

JohnTheCoolingFan, 2024-07-14:
Now I'm getting really weird layer shifts, in this print they are on the X axis. Previously they were on the Y axis, and this is unrelated to backlash compensation, as it was put into firmware. I have basically all components and models to start an upgrade of the printer, but it suddenly decided to not produce usable prints anymore. Very weird...

[image]
[image]
[image]

Rixxxi, 2024-07-14:
Those layer shifts are really anoying. I hope we can get them sorted out finally one day.

> this is unrelated to backlash compensation, as it was put into firmware.
What do you mean?

JohnTheCoolingFan, 2024-07-14:
My issues started appearing after I've disassembled the Y axis, there were no layer shifting issues with the firmware I used before the disassembly and reassembly.
I've put the backlash compensation config values into the firmware (so instead of gcode I put it in the Marlin Configuration file)

Rixxxi, 2024-07-14:
> I've put the backlash compensation config values into the firmware (so instead of gcode I put it in the Marlin Configuration file), which avoids the issue.

Thanks. I wasn't aware that Configuration file vs. GCODE makes any difference. Is this a known problem?
However, I had my backlash compensation in the Configuration file all the time. So it won't make a difference for my layer shift problems.

schmttc, 2024-07-15:
@Rixxxi I'm not actively using this printer any more, but:

>My understandig is that 2.1.2.4 should have a the fix for the Layer Shifts issue https://github.com/schmttc/EasyThreeD-K7-STM32/issues/2 included. Is that correct?

From what I can see yes this fix should be in main Marlin now, it was merged to bugfix in Nov 23 and there's been a few releases since then. You can compare the code to confirm.

>I have not enabled BACKLASH_SMOOTHING_MM so far. Is it recommended to do so?

Smoothing averages out the backlash over a longer distance, which gives smoother but less precise results. I haven't done any testing with this feature.

> Can you explain a little more about what you found out on the acceleration? Which acceleration settings work, which don't? Is it simple as "lower acceleration is always more safe then higher" or are there particular acceleration settings that will/won't work?

I used two test scenarios, 20mm/s/s and 100mm/s/s, but didn't do any tuning to figure out where the upper limit of 'safe' is.
But then when the backlash correction fix was implemented, I could use 100mmss accelerations again, but didn't do any extensive testing.
But based on John's and your results, it looks like there is still an acceleration limit with recent firmware and these printers. If in your testing 30mmss is safe, but 40mmss isn't, that aligns with my results and gives us a better data point that 30mms is the safe limit.

Why the newer firmware only works with low acceleration but the older firmware seems to work with higher acceleration is a question I don't know the answer to. It would probably need lots of testing between new and old and probably isn't worth the time given I found the linear print speed is limited to 40mm/s to get quality prints.

Rixxxi, 2024-07-15:
@schmttc thanks for getting back on my questions.

> You can compare the code to confirm.

I will do so.

> But based on John's and your results, it looks like there is still an acceleration limit with recent firmware and these printers. If in your testing 30mmss is safe, but 40mmss isn't, that aligns with my results and gives us a better data point that 30mms is the safe limit.

The value of 30 was in my case the print speed ("feed rate") in mm/s and not the acceleration (mm/s²). I was wondering if this could be a limitation of the 264-motors. As their gearbox has a higher reduction compared to the 586-motors, I could imagine that only 586 motors are able to print with 40mm/s and the 264-motors probably don't. But I couldn't find any information about this so far.

> Why the newer firmware only works with low acceleration but the older firmware seems to work with higher acceleration is a question I don't know the answer to. 

Can you tell a firmware version that shall be safe to print with "high" acceleration? What acceleration value did you use with it? Can you tell the type of motors (type 264 = ca. 467 steps // type 586 = ca. 606 steps) your printer is based on?

schmttc, 2024-07-23:
- Apologies for the mix up. I print at 40mm/s speed, which ensures a good quality print. I tried 60mm/s but the print quality suffered in general. And I tried printing at 120mm/s in vase mode when testing the Input Shaping feature, it worked ok but sometimes had layer shifts
- Checking Aliexpress, is seems the 264 motor has a 32x reduction, and the 586 motor has a 64x reduction. My intuition is that a bigger reduction has more inertia in the geartrain and so would cause more problems at higher speed or acceleration
- From testing I think the layer shifts are more related to acceleration than speed
- The firmware that worked with 'high' acceleration is the original v1.1 firmware :) I can't recall what settings I used but probably the firmware default 300mms2

[21 more replies]

PR #11 (2023-12-20): K9 version
This pull request adds compatibility for the K9 printer using ET4000+ board

Closes #10 

Also the PR discussion will serve to log the progress of exploration of the device.
closed 2024-09-20 by schmttc