schmttc/EasyThreeD-K7-STM32 #12: Layer Shifting on 2.0.9.1

OddFrog, 2023-10-06:
Recently got a K7 with the leveling buttons and an ET4000+. I Flashed 2.0.9.1 and I was getting layer shifting so I flashed 2.1.2 and the shifting is resolved but it still has the backlash fix enabled. I ran M503 and M115 that reported marlin 2.1.2 and backlash compensation set at F1.00 S0.00 X0.20 Y0.36 Z0.00 I attached the outputs as a text file.
[2.1.2 test.txt](https://github.com/schmttc/EasyThreeD-K7-STM32/files/12835589/2.1.2.test.txt)
Really Seems i have the best of both Right now but it doesn't track with what the descriptions on the releases page says.

schmttc, 2023-10-07:
Someone over in #2 found the same thing, no layer shift with 2.1.x. I also had some prints work, but specifically the backlash test tower I use causes the issue to reappear. 
Also, setting acceleration low (at least 20, but somewhere less than 100mm/s2) causes the problem to go away.
But if it's working for you, then that's good :)

You can also try the '[backlashfix](https://github.com/schmttc/EasyThreeD-K7-STM32/tree/backlashfix)' branch, I've implemented a fix for the layer shift that is working for me, and there's also a binary with input shaping enabled if you're interested.

OddFrog, 2023-10-07:
I'll give the input shaping a try! Thank you for working on this you're the only reason I even considered buying the printer to begin with and have been having a blast tinkering with it!

schmttc, 2023-11-06:
I'm glad you're having fun with it. With the backlash issue closed out (#2) they're a half decent little printer as long as you are aware of the limitations.
The next Marlin release will have the backlash fix, I have this branch in the mean time https://github.com/schmttc/EasyThreeD-K7-STM32/tree/backlashfix

closed 2023-11-06 by schmttc