schmttc/EasyThreeD-K7-STM32 #15: Enhancements for boards with AT32F403ARCT7

Vexisu, 2024-01-11:
Hi,
a couple days ago I've received a K9 with AT32F403ARCT7 and now I wonder if there is an option for compiling Marlin for Cortex-M4F arch and 240MHz clock while using STM32F103 config. I'm not familiar with PlatformIO, but used STM32 HAL libraries, therefore I leave my thoughts here for consideration, and with some help I may be able to contribute to this project.

PS: Thanks for y'all for supporting this niche printers. After short time of testing I can say, that this firmware works much better that original one.

kossnikita, 2024-01-16:
I tried to create a configuration for Cortex-M4, but immediately found several problems with porting. For some reason Marlin does not support disabling JTAG on the M4, although it is required on the ET4000+. Having problems with pin names, I didn’t spend much time on it and gave up.

If you are able to successfully build the firmware, please try to enable I2C as well. See https://github.com/schmttc/EasyThreeD-K7-STM32/issues/5#issuecomment-1883304759

kossnikita, 2024-01-16:
According to Artery cross reference table, you may be intrested in STM32F446

[image]

platformio has [genericSTM32F446RC](https://docs.platformio.org/en/stable/boards/ststm32/genericSTM32F446RC.html) board

Vexisu, 2024-01-17:
After some thought and tinkering, I've got to the conclusion that it may be better to only change a source clock to HSI and increase the CPU frequency. Porting to other arch is especially difficult, because rn this board only works with libmaple.

Vexisu, 2024-01-20:
Found some time to dig into the topic a little deeper. With libmaple it should be possible to easily define a new board and cpu variant, and then to increase the cpu clock to 120 MHz. At this moment I'm not sure what it will take to get the mcu working at full speed, as registers for PPL configuration may differ between STM32F103 and AT32 (libmaple doesn't have macro definitions for higher settings).

kossnikita, 2024-01-21:
AT32F403A has extra PLL config bits that STM32F103 doesn't have
[image]

Amax-prog, 2024-07-05:
Hello. I have a K7 printer with an ET4000 board. I downloaded the firmware from the main branch into it, but its start button stopped working. When you press the start button, printing does not start, and when you hold it, the extruder does not rise. If you download the standard firmware from the manufacturer, then everything works. I tried downloading different firmware for MKS Robin Lite, but the result is the same - the start button does not work. Do all the buttons on the K7 ET4000+ printer work with the firmware from the main branch or with the EasyThreeD-K7-STM32-ET4000PLUS-K7-rev-B firmware? Maybe someone has uncompiled firmware from the manufacturer, I want to add an OLED display for the printer. Also, when connecting the display, the files on the SD card (32gb fat32 mbf) are not displayed, but the files are displayed on the Biqu printer. Has anyone encountered this?

Vexisu, 2024-07-05:
@Amax-prog does your printer have leveling buttons? In this case use rev. B firmware, as the connections for buttons are different. If it won't work, create a new issue on this repository and provide photos of the mainboard. You can download the firmware directly from my fork (mksLite.bin)