schmttc/EasyThreeD-K7-STM32 #20: The print button does not work

Amax-prog, 2024-07-06:
Hello. I have a K7 printer with an ET4000+ board. I downloaded the firmware into it from the main branch, but its start button did not work. When the start button is pressed, the ignition does not start, and when held, the extruder does not rise. If you download the standard firmware from the manufacturer, then everything works. I tried to download different firmware for the ISS Robin Light, but the result was the same - the start button does not work. Do all the buttons on the K7 ET4000+ printer work with the firmware from the main branch or with the EasyThreeD-K7-STM32-ET4000PLUS-K7-rev-B firmware? Maybe someone has uncompiled firmware from the manufacturer and wants to add an OLED display to the printer. Also, when connecting the display, the files on the SD card (32 GB fat32 MBF) are not displayed, but the files are displayed on the Biqu printer. Has anyone encountered this?
[image]
[image]

Amax-prog, 2024-07-07:
I was able to solve the problem. You need to compile the firmware not for MKS Robin Lite, but for MKS Robin Lite Maple

schmttc, 2024-09-19:
Updated main branch to set MKS Robin Lite Maple as the default platform.

closed 2024-09-19 by schmttc