schmttc/EasyThreeD-K7-STM32 #22: newbie asking for guidance

astemiolup1n, 2024-07-21:
hi to everyone, and a special thanks for all your effort. i'm really new to this word and i recently bought a k7. i noticed squared circle issue on standard firmware. is this still operating? latest update does work properly ? how do i install it? is just a matter of swap mksLite.bin with the original in the sd? or i should do some correction? i have to tell in advance that i'm not so confident with programming or compile XD

schmttc, 2024-07-23:
Hello!

Yes, the backlash correction feature in this firmware fixes the squared circle issue. 

I'm slowly working on a user guide, please see the install instructions here
https://github.com/schmttc/EasyThreed-Community-Firmware/wiki/Beginner%E2%80%99s-Guide#5-install-the-firmware

Use the binary from the main repo https://github.com/schmttc/EasyThreeD-K7-STM32/raw/master/mksLite.bin

And let us know if you run into any issues :)

astemiolup1n, 2024-07-23:
thanks for your reply 
my k7 have 4000+ board
opening my sd this are the file that i found 
[image]
i can only see mks_pft.sys no mksLite.cur to take for a recovery

schmttc, 2024-07-23:
If the board is ET4000+ and your printer does not have the 4 levelling buttons, you can find a copy of the original firmware for backup here https://github.com/schmttc/EasyThreeD-K7-STM32/tree/Original-Firmware-Marlin-1.1.1/Default%20Firmware

jonylentzmc, 2024-07-30:
Sorry for this kind of question, but what do I need to compile the FW for my printer? Just the FW source code, PlatformIO and the modified files section? Do I need to replace anything else for the FW to work with this printer? (ET4000+ - Dog [K9] version)

I'm currently using [https://github.com/schmttc/EasyThreeD-K7-STM32/issues/10#issuecomment-2101984184](this compiled version) as it fixed the EEPROM being written to SD card and the physical buttons

I'm trying to mod the printer with different print area sizes, endstops switches and other modifications, so I'll need to change a couple of settings in the configuration.h and configuration_adv.h thus I need to recompile the FW...

schmttc, 2024-08-02:
@jonylentzmc no stress
@JohnTheCoolingFan added in a Github auto binary build action, which I've just applied to the ET4000PLUS-K9 branch.

So if you fork this repo in Github, you can make your own modifications on ET4000PLUS-K9 and when you push the changes in a few minutes the new binary will be there (check Actions for status)

Otherwise yes the old fashioned way is PlatformIO and the source code for the ET4000PLUS-K9 branch :)

jonylentzmc, 2024-08-08:
> @jonylentzmc no stress @JohnTheCoolingFan added in a Github auto binary build action, which I've just applied to the ET4000PLUS-K9 branch.
>
> So if you fork this repo in Github, you can make your own [...]

Thank you so much for the reply : )

closed 2024-09-17 by schmttc