schmttc/EasyThreeD-K7-STM32 #28: Layer Shift Issues

schmttc, 2024-09-19:
Hi Everyone,

After closing #2, a number of people have continued to have layer shift issues under various circumstances. I thought that maybe this was due to acceleration or jerk settings, but using the K7 firmware in main branch 179b6f3 I successfully printed a backlash tower at 80mm/s print speed and 1000mm/s2 acceleration with backlash compensation enabled and had no layer shift.
I understand that this small model doesn't reach up to these limits, but in resolving #2 at print speed 40mm/s I could alter acceleration to the point that layer shift did not occur (20mm/s2) or did occur (>~100mm/s2). By massively exceeding these limits, I have concluded that backlash compensation is not causing backlash, nor is acceleration. 

So if you still have issues with layer shift, please include full details, including:
- Printer model
- Motherboard model
- Specific version of this firmware you're trying
- If you have customised the firmware, 
- If you have customised your printer hardware
- Print settings (preferable a project file)
- Link to the model 

This way we can attempt to reproduce the issue and resolve the root cause.

**Print Artefacts**
[Backlashtower_0_20mm.80x1000.3mf.txt](https://github.com/user-attachments/files/17053457/Backlashtower_0_20mm.80x1000.3mf.txt)
[Backlashtower_0_20mm.80x1000.gcode.txt](https://github.com/user-attachments/files/17053469/Backlashtower_0_20mm.80x1000.gcode.txt)
[image]

schmttc, 2024-09-19:
Issue: K7 no layer shift printing at 60mm/s with 1000mm/s2 acceleration with default firmware.
Set higher max speed and acceleration in start G-code to override firmware defaults
[image]

Printer model: K7
Motherboard model: ET4000+
Specific version of this firmware you're trying: config/EasyThreeD/ET4000PLUS/K7/mksLite.bin from 179b6f3
If you have customised the firmware: No
If you have customised your printer hardware: Cooling duct
Print settings (preferable a project file): [3dbenchy.60x1000.3mf.txt](https://github.com/user-attachments/files/17067778/3dbenchy.60x1000.3mf.txt)
Link to the model: in project file

Rixxxi, 2024-10-25:
Hi there,

My setup:
- Printer model: EasyThreeD X1
- Motherboard model: ET-4000 V2
- Specific version of this firmware you're trying: Multiple
- If you have customised the firmware: #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
- If you have customised your printer hardware: installed the [RepRapDiscount Full Graphic Smart Controller](https://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller)
- Print settings (preferable a project file): Attached
- Link to the model: [The Penguin](https://www.thingiverse.com/thing:22981)

I'm suffering from layer shifts for a long time. Here are some of my previsous posts offering more details about my "journey": [post 1](https://github.com/schmttc/EasyThreeD-K7-STM32/issues/10#issuecomment-2213746761), [post 2](https://github.com/schmttc/EasyThreeD-K7-STM32/issues/10#issuecomment-2254562332), [post 3](https://github.com/schmttc/EasyThreeD-K7-STM32/issues/10#issuecomment-2328706935)

Over the last days I did some more testing regarding my layer shifts. **I always related the layer shifts to the firmware. Spoiler: It's probably NOT!** Here is a wrap up of what I tested:
- Various firmware versions and configurations, including: from [DaDuke](https://daduke.org/hardware/X1/), original Marlin 2.1.2, Marlin 2.1.2.4, Marlin 2.1.3.0 (unreleased), schmttc_EasyThreeD-K7-STM32-master
- alternate the print speed and acceleration (down to 20mm/s²)
- [(re)tested](https://github.com/schmttc/EasyThreeD-K7-STM32/issues/10#issuecomment-2254562332) my printers maximum speed and acceleration (result: 55mm/s at 400mm/s²) and never printed with more than 40mm/s (most tests at 30mm/s or slower)
- disabled the RepRap LCD (idea: save MCU performance)
- used different microSD
- used different slicer (Orca, Cura)
- disabled backlash compensation in firmware ( ...and painfully realized: **layer shifts still occured**!)
Now I realized it might not be firmeware related:
- tightened the belt tension
- checked my Vref voltage. It was about ~1.45V which relates to 0.9A stepper current and is quite high. I read that a too high current can make the stepper driver run hot and run into protection mode. My stepper drivers don't have any cooling (not even passive) and using my finger the X and Y drivers felt quite hot (could hold my finger on it for 3-4 seconds). So I lowered Vref to ~1.1V (about 0,7A)

Nothing helped!!! This is so frustrating :-((((
Well, as far as I remember printing at only 20mm/s² acceleration worked. I will test that again. But it wouldn't be a satisfying solution.
I'm running out of ideas what I could test. Any ideas?

Thanks,
Rixi

Rixxxi, 2024-10-25:
This afternoon I did another test print (penguin) with 20mm/s² acceleration (sliced to 20mm/s² with Orcaslicer). It actually prints without errors. It just takes 2.5-3 times as long as with normal acceleration.

So not really satisfying, especially as I tested the ‘machine limits’ via Proterface a few weeks ago. I was able to move the X and Y axes at up to 55mm/s at 400mm/s² without any problems. So in my opinion it can't be due to the printers mechanics that it only prints flawless at 20mm/s² acceleration.