schmttc/EasyThreeD-K7-STM32 #32: Firmware does not update from SD card

billykindell, 2025-01-04:
Hello, I'm really hoping that this is still actively maintained as I would love to get involved. The only problem is that at the moment, with just "mksLite.bin" on the SD card, it does nothing. The printer (K9) just stays in a powered on state and never installs the firmware. The board is confirmed to be ET4000+ so I am very confused.

Any help would be incredible. Thank you.

Ajimaru, 2025-01-05:
> Hello, I'm really hoping that this is still actively maintained as I would love to get involved. The only problem is that at the moment, with just "mksLite.bin" on the SD card, it does nothing. The [...]

@billykindell  Can you checkt this [https://github.com/schmttc/EasyThreeD-K7-STM32/issues/10#issuecomment-2008200686](https://github.com/schmttc/EasyThreeD-K7-STM32/issues/10#issuecomment-2008200686) ?

billykindell, 2025-01-05:
@Ajimaru So I will need a copy of that EEPROM.DAT file? my printer doesn't create this, so I guess I will have to look for a dump somewhere?

billykindell, 2025-01-05:
I'm actually at a bit of a loss. I can't print anymore, the SD card works fine on windows and i can see files there, but the printer is acting like there is no SD card inserted.

schmttc, 2025-01-05:
We fixed the EEPROM.dat issue a little while ago, the firmware now uses onboard flash for storage

@billykindell what format did you use for the SD card? This indicates that it needs to be FAT32
https://github.com/makerbase-mks/MKS-Robin/issues/66#issuecomment-441505545

billykindell, 2025-01-05:
I've tried formatting as FAT, EX FAT and FAT32. With various sizes allocated and now tested on 3 more SD cards. None appear to be working

billykindell, 2025-01-07:
Still no joy. I've no idea what happened but it will not read any SD card since initially trying to flash firmware. User error no doubt but it's now only functional via usb.

I'm probably going to replace the main board, but with a bigtreetech mini V3, and just use this firmware.

schmttc, 2025-01-07:
Over USB try playing with the SD commands, M21 should try and mount the card. Maybe autodetect isn't working, or you might get an error to follow up.

https://marlinfw.org/docs/gcode/M021.html

billykindell, 2025-01-08:
> Over USB try playing with the SD commands, M21 should try and mount the card. Maybe autodetect isn't working, or you might get an error to follow up.
> 
> https://marlinfw.org/docs/gcode/M021.html

Ah excellent thank you, it just returns "volume.init failed" on every card so I will just use USB until the BTT board arrives

billykindell, 2025-01-08:
This was really bugging me because I hate not knowing what the cause is. Turns out all of my SD cards were formatted as GPT drives. I changed the initial "broken" card to MBR and now it works fine.

billykindell, 2025-01-09:
I have _another_ SD card on order to try and restore to original firmware, but I do also have a BTT E3 board coming which I'm hoping I can make work with the original hardware.